SiliconLabsSoftware/arduino #112: [bug]  String.cpp: undefined reference to ltoa and ultoa

Guswib, 2025-02-02:
### Hardware

Arduino Nano Matter

### Core version

2.20

### Arduino IDE version

2.3.4

### Operating system

Windows 10

### Radio stack variant

No radio

### Issue description

During compilation in the arduino IDE, the following error message is given:

<user>/appdata/local/arduino15/packages/siliconlabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: <user>\AppData\Local\arduino\sketches\D54C9E1FF4DB4B8DD7CB06F3C73E69AA/core\core.a(String.cpp.o): in function `arduino::String::String(long, unsigned char)':
<user>\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\2.2.0\cores\silabs\api/String.cpp:113: undefined reference to `ltoa'
<user>/appdata/local/arduino15/packages/siliconlabs/tools/gcc-arm-none-eabi/12.2.rel1/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: <user>\AppData\Local\arduino\sketches\D54C9E1FF4DB4B8DD7CB06F3C73E69AA/core\core.a(String.cpp.o): in function `arduino::String::String(unsigned long, unsigned char)':
<user>\AppData\Local\Arduino15\packages\SiliconLabs\hardware\silabs\2.2.0\cores\silabs\api/String.cpp:121: undefined reference to `ultoa'

### Minimal reproducer code

void setup() {
  // put your setup code here, to run once:
  long l_num=10;
  String(l_num,10);
  unsigned long ul_num=10;
  String(ul_num,10);
  
 
}

void loop() {
  // put your main code here, to run repeatedly:

}

leonardocavagnis, 2025-02-27:
Dear @Guswib,
Thanks for reporting this issue.

Currently, the `itoa` library is missing. 😞

If you need it, you can copy and paste [this itoa.c](https://github.com/arduino/ArduinoCore-renesas/blob/8a0bbaa388406ba3a727850ab09be670d375dad0/cores/arduino/itoa.c) implementation into the `core/silabs` folder (but remember to remove this line: #include <api/deprecated-avr-comp/avr/dtostrf.c.impl>).

If you'd like to contribute, feel free to open a PR with this fix!

Guswib, 2025-03-06:
Dear @leonardocavagnis,

I will have a look andn hope to find a solution. 

Cheers
G

Guswib, 2025-03-06:
Hi, 
The cloning of the repo does not work:
_______________________________________________________________________________________________
Cloning into 'C:\Users\gusta\Documents\GitHub\CoreFIX\arduino2'...
remote: Enumerating objects: 6405, done.        
remote: Counting objects: 100% (1157/1157), done.        
remote: Compressing objects: 100% (660/660), done.        
remote: Total 6405 (delta 524), reused 992 (delta 441), pack-reused 5248 (from 1)        
Receiving objects: 100% (6405/6405), 47.02 MiB | 1.65 MiB/s, done.
Resolving deltas: 100% (2880/2880), done.
Updating files: 100% (19279/19279), done.
Downloading variants/bgm220explorerkit/ble_silabs/gsdk.a (3.4 MB)
**Error downloading object: variants/bgm220explorerkit/ble_silabs/gsdk.a** (d1cd1d1): Smudge error: Error downloading variants/bgm220explorerkit/ble_silabs/gsdk.a (d1cd1d13de51480e6c7a8d4346b3d073ae5fafa23a0a9ad8621fdbbb8cc2a2fc): [d1cd1d13de51480e6c7a8d4346b3d073ae5fafa23a0a9ad8621fdbbb8cc2a2fc] **Object does not exist on the server: [404] Object does not exist on the server**

Errors logged to 'C:\Users\gusta\Documents\GitHub\CoreFIX\arduino2\.git\lfs\logs\20250306T160839.2974214.log'.
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: variants/bgm220explorerkit/ble_silabs/gsdk.a: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'

""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""""
This error also occures  but for other .a files

silabs-bozont, 2025-04-08:
Hello @Guswib,
Thank you for the bug report! It seems like @leonardocavagnis was right - we're missing the implementation of `itoa` `ltoa` `utoa` and `ultoa`.
I've added them - they'll be included in the next release.

About the cloning issue - it seems like the LFS artifacts haven't been published correctly - I'll look into putting them out. They're in the release archives but seems to be missing from git.

silabs-bozont, 2025-07-28:
Hello @Guswib,
This issue has been fixed by the latest (3.0.0) release.

closed 2025-07-28 by silabs-bozont