SiliconLabsSoftware/arduino #118: DeepSleepExternalOrTimeWakeup.ino, wakes up immediately without DeepSleep for 10 seconds

msfujino, 2025-02-26:
I am trying “DeepSleepExternalOrTimeWakeup.ino” but it wakes up immediately without DeepSleep for 10 seconds.
I think there is a problem with LINE130 ~ LINE134 in "ArduinoLowPower.cpp".

grobasoz, 2025-02-27:
It could be because the same pin PC0 is used for the Wakeup as well as the Escape Pin for programming. I used a different Wake Pin (PA5 = D10) and it works fine.

msfujino, 2025-02-27:
I tried changing the pin, but it just wakes up when I put my finger **near** the pin.
It does not appear to be pulled down.

grobasoz, 2025-02-27:
I suggest you initialise the pin to either pullup or pulldown. I use PA5 as it can be used to wake from sleep modes using the attachInterrupt function.

```c
  // Running IRQ Pin
  pinMode(WAKE_UP_PIN, INPUT_PULLUP);
  attachInterrupt(WAKE_UP_PIN, myWakeup, FALLING);

  // Set up the wake-up pin
  LowPower.attachInterruptWakeup(WAKE_UP_PIN, nullptr, FALLING);
```

msfujino, 2025-02-27:
This problem does not occur with ExternalWakeup; it occurs when TimedWakeup and ExternalWakeup are used at the same time.
I think there is a problem with LINE130 ~ LINE134 in "ArduinoLowPower.cpp".

grobasoz, 2025-02-28:
I'm using the example you mentioned, viz DeepSleepExternalOrTimedWakeup.ino...

grobasoz, 2025-02-28:
I tested your code and it worked fine. At 11:06:54 I triggered a Pin Wake (PA5).

11:06:28.091 [10.657] RX> Deep sleep with external or timed wakeup
11:06:29.071 [00.979] RX> Going to deep sleep for 10s at 1037
11:06:39.735 [10.663] RX> Deep sleep with external or timed wakeup
11:06:40.716 [00.980] RX> Going to deep sleep for 10s at 1037
11:06:51.380 [10.662] RX> Deep sleep with external or timed wakeup
11:06:52.361 [00.979] RX> Going to deep sleep for 10s at 1037
_11:06:54.852 [02.490] RX> Deep sleep with external or timed wakeup_
11:06:55.833 [00.979] RX> Going to deep sleep for 10s at 1037
11:07:06.497 [10.662] RX> Deep sleep with external or timed wakeup
11:07:07.479 [00.980] RX> Going to deep sleep for 10s at 1037
11:07:18.142 [10.661] RX> Deep sleep with external or timed wakeup
11:07:19.122 [00.978] RX> Going to deep sleep for 10s at 1037

msfujino, 2025-02-28:
It's strange. My XIAO_MG24 is floating in the air with nothing connected to its pins except it is connected to the PC via USB-C so that it is not affected by its surroundings.

I am wondering why LINE130-134 in ArduinoLowPower.cpp has a different command order compared to LINE102-108. If I change the order to be similar to LINE102-108, the problem will not occur.

grobasoz, 2025-02-28:
OK - I see now. You want to enable Pin Retention on exit from EM4? 
Yes, I suppose that would be necessary if the pin was intended to be just left floating...
In my testing, I had the PA5 pin set to a defined state - either High or Low, never just floating.

msfujino, 2025-02-28:
A mechanical switch was assumed for the wake-up signal.