SiliconLabsSoftware/arduino #119: [bug] <The board Seeed Studio XIAO MG24Sense does not support>

Jasionf, 2025-02-27:
### Hardware

Seeed Studio XIAO MG24 Sense

### Core version

2.0.0

### Arduino IDE version

2.3.0

### Operating system

macOS

### Radio stack variant

BLE

### Issue description

I am seeed stduio xiao mg24 sense to use the official ble_hid_mouse.ino example when compiling seems to be an error, and does not support XIAO mg24 sense board, is there any other operation can make the compilation pass!

### Serial output

[image]

leonardocavagnis, 2025-02-27:
Dear @Jasionf,
The XIAO MG24 Sense board uses a different IMU model (LSM6DS3), so this sketch won’t work as is.
You'll need to modify it using the LSM6DS3 library. More details can be found [here](https://wiki.seeedstudio.com/xiao_mg24_sense_built_in_sensor/#xiao-mg24-sense-imu).

Jasionf, 2025-02-27:
I am an application engineer at Seeed Studio, and I recently updated the onboard package for our XIAO MG24 Sense. While exploring the available examples on Arduino, I noticed that the IMU sensor used in our device is not currently supported.

I would like to inquire if there are any plans to support our IMU sensor model in the future. Your guidance on this matter would be greatly appreciated, as it would greatly enhance our users' experience with the XIAO MG24 Sense.

Thank you for your attention, and I look forward to your response.

silabs-bozont, 2025-02-27:
I have some excellent news for you - we have a BLE HID Mouse example just for the XIAO MG24 right next to the other one ;) 
https://github.com/SiliconLabs/arduino/blob/main/libraries/SiliconLabs/examples/ble_hid_mouse_xiao_mg24/ble_hid_mouse_xiao_mg24.ino
Give it a spin!

Jasionf, 2025-02-27:
Oh, I noticed it. Thank you very much.