SiliconLabsSoftware/arduino #123: [bug] Wrong Pins assigned to Serial1

Prasana-sys, 2025-04-28:
### Hardware

xG24 Explorer Kit

### Core version

2.2.0

### Arduino IDE version

2.3.5

### Operating system

Windows 11

### Radio stack variant

BLE

### Issue description

In pins_arduino.h for the xG24 Explorer Kit, Serial1 pins are defined as:

```
#define PIN_SERIAL_RX1 (A7) // PA0
#define PIN_SERIAL_TX1 (A6) // PB1
```

https://github.com/SiliconLabs/arduino/blame/c51e6c01f2e90f46d08e6d530a015531050cc42c/variants/xg24explorerkit/pins_arduino.h#L85

I believe this is wrong and it should be RX1 -> A5 (PD5) and TX1 -> A4 (PD4) as listed on the datasheet and from what I can see physically on the board. Please reconfirm this.

[image]

silabs-bozont, 2025-04-29:
Hello @Prasana-sys,
Thank you for your bug report! According to the original manual and the MikroBUS header pinout you're right - the `Serial1` pins should be at `PD4` and `PD5`.
However to fit all the peripherals we had to make some puzzle moves. We added an additional SPI (`SPI1`) which is shared with `Serial1` using the `EUSART0` peripheral - which doesn't allow mapping to `PortC` and `PortD` pins.
Right now the xG24 Explorer Kit's EUSART/USART peripheral mapping looks like this:
- USART0 -> Serial (USB)
- EUSART0 -> Serial1 or SPI1
- EUSART1 -> SPI

I'll look into whether it's possible to shuffle the peripherals around to get the correct mapping for `Serial1` in order to not break the MikroBUS pinout. This'd also mean that the `SPI1` has to move with `Serial1` as well.

Here's the current Arduino pinout for reference:

[image]

silabs-bozont, 2025-05-13:
Hello @Prasana-sys,
I've remapped the USART/EUSART peripherals on the xG24 Explorer Kit to match the mikroBUS pinout. It was possible to swap around the USART/EUSART peripherals to have the correct layout.
The new USART/EUSART mapping for the xG24 ExpKit is:
- USART0 -> SPI
- EUSART0 -> Serial (USB)
- EUSART1 -> Serial1, SPI1

Here's the updated pinout:
[image]

Changing this opened a bigger rabbit hole than I expected - a few adjustments were also needed in the SPI driver - but thankfully everything works as expected with the new layout as well.

I also changed the pin mapping on the _Ezurio Lyra 24P_ - which was having a similar mapping issue.
The changes will be included in the next release!

silabs-bozont, 2025-07-28:
Hello @Prasana-sys,
This issue has been fixed by the latest (3.0.0) release.

closed 2025-07-28 by silabs-bozont