SiliconLabsSoftware/arduino #131: [bug] EEPROM doesn't work on BLE stack for MG24

wwww-wwww, 2025-06-21:
### Hardware

Seeed XIAO MG24 Sense

### Core version

2.3.0

### Arduino IDE version

2.3.6

### Operating system

Windows 11

### Radio stack variant

BLE

### Issue description

EEPROM.read always returns 255 when not using Matter stack (BLE (Arduino), BLE (Silabs), None)

silabs-bozont, 2025-06-23:
Hey @wwww-wwww,
Thanks for your report!
I've just re-ran our EEPROM tests on the XIAO MG24 which were all successful in writing then verifying all of the EEPROM bytes on all the radio variants. I'm not observing any issue with the EEPROM library on my end.

Could you please attach a minimal reproducer sketch? I could try and reproduce the issue on my end.
You could also try re-burning the bootloader - which will re-format the EEPROM as well - in case there was any corruption.
Let me know how it went!

wwww-wwww, 2025-06-26:
My test was
```c++
void setup() {
  Serial.begin(115200);
  EEPROM.write(0, 0);
  EEPROM.write(1, 1);
  EEPROM.write(2, 2);
  EEPROM.write(3, 3);
  Serial.printf("%d %d %d %d\n", EEPROM.read(0), EEPROM.read(1), EEPROM.read(2),
                EEPROM.read(3));
}
```

But re-burning the bootloader seemed to have worked. 🤷 Thanks

closed 2025-06-26 by wwww-wwww