SiliconLabsSoftware/arduino #135: [request] xiao mg24 provisions

vincenttl, 2025-07-24:
### Summary

Is it possible to add the xiao mg24 to
 arduino_matter_provision.py 

Or let me know how to do it. Ill do the contribution

closed 2025-07-28 by silabs-bozont

vincenttl, 2025-07-28:
Works well, thank you. 
Can we add instructions on how to create more than 3 configs?

silabs-bozont, 2025-07-31:
Glad it works well! Good idea, we'll add instructions to the readme on how to generate these binaries, but I can summarize it here for you quickly.

I'm using our Matter SDK's [provisioning tool](https://github.com/SiliconLabs/matter/tree/release_2.3.1-1.3/provision) to generate provisioning data on the devices. 

This tool only support J-Link devices (the XIAO and the Nano Matter has openOCD) - so I'm using a [Silicon Labs Wirleless Pro Kit](https://www.silabs.com/development-tools/wireless/wireless-pro-kit-mainboard?tab=overview) mainboard with an add-on radio board which has the identical chip as the XIAO/Nano.
For the XIAO (EFR32MG24B220F1536IM48-B) it's the: [xG24-RB4187C](https://eu.mouser.com/ProductDetail/Silicon-Labs/xG24-RB4187C?qs=MyNHzdoqoQLnCpBvKq5fEw%3D%3D).

After running the provisioning tool with the Pro Kit - I dump the NVM3 of the chip to a hex file - which will be the provisioning binary that the script uses. You can see the provisioning tool parameters I used for each binary in the script's source.

vincenttl, 2025-07-31:
Thanks. I was expecting something like that. So without a board that supports jlink it's very difficult to generate more hex dumps.