SiliconLabsSoftware/arduino #145: [request] Add Credential/PIN capabilities for Matter Door Lock

andreask1, 2025-08-30:
### Summary

The Matter Door Lock example and library currently do not support setting a credential/pin. 

Without a pin, unlocking via voice on a smart speaker / display (e.g., Google Nest Hub) is not possible.

https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/examples/matter_door_lock/matter_door_lock.ino
https://github.com/SiliconLabs/arduino/blob/main/libraries/Matter/src/MatterDoorLock.cpp

annoscholten, 2025-11-03:
@silabs-bozont  - would you be able to share a schedule for this feature request?

lboue, 2026-09-03:
@annoscholten What kind of keyboard would you like to use to enter the PIN?

lboue, 2026-09-04:
- https://github.com/SiliconLabsSoftware/arduino/issues/198